joshajohnson/sea-picro #10: v.03 test

freznel10, 2022-07-25:
Hi Josh, has v.3 been tested already?

joshajohnson, 2022-07-26:
Hey Freznel, it''s partially tested.

I've checked all the IO work, but need to check VBUS detection, RGB LED, check a few voltages, and a few other things.
If you can hold off for 48 hours I'll have finished testing and update here with (hopefully) good news.

joshajohnson, 2022-07-27:
Hi @freznel10,

I assembled a few more boards and tested them, have found two more issues documented in https://github.com/joshajohnson/sea-picro/issues/11.

Will hopefully resolve by this time tomorrow assuming all goes well.

joshajohnson, 2022-07-28:
All tested now @freznel10!

I'm ordering a bunch of the 1.0 boards now, so fingers crossed there are no more bugs!

closed 2022-07-28 by joshajohnson

freznel10, 2022-07-28:
Nice to hear that!:) Hope everything turns out fine for the 1 boards!