joshajohnson/sea-picro #12: How does the reset/boot circuit work?

girishji, 2022-09-27:


joshajohnson, 2022-09-27:
Hi,

The README explains its function per below.

> When tapped for less than 500ms will reset the board, but when held for 500ms or more will jump into the bootloader.

You can also find the LTspice simulation in the simulation folder.-------- Original Message --------

girishji, 2022-09-28:
I understand the usage. Is there a write up somewhere (reference) how this circuit works in principle? It is not intuitive how these mosfet switches cause delay.

joshajohnson, 2022-09-28:
R2, R3, and Q2 are to invert the active low signal of a pro micro for backwards compatibility. I have since realised that Q2 and R2 can be removed and SW1 placed where Q2 is.

R1 pulls nRST (RUN) high to enable the chip, and Q1 pulls it low to got into reset when Q2 turns off / SW1 is pressed.

For the delay circuit we need Q3 as the BOOT pin is also flash chip select, so if we had capacitance on that line the flash would never work.
R4 is probably not needed, but I never like pulling anything directly to ground.
The delay circuit could be made with 3 parts instead of 5, but I wanted to keep BOM lines low.

Q4 (could sub for a diode) is to ensure the gate of Q3 stays charged once you release the button - ensuring reset comes up before BOOT.

girishji, 2022-09-29:
When SW1 is not pressed you have 3v3 connected to 5k+3*51k resistors, resulting in 20 ma current drained continuously, no?
Also, is the RC delay circuit tested with components (not simulation)? RC delay is a good idea for this application.

joshajohnson, 2022-09-29:
I've manufactured and tested over 100 boards, so yes it's proven in real life.

It draws 600uA in the unpressed state.

closed 2022-12-07 by joshajohnson