joshajohnson/sea-picro #15: Sea Picro maybe bricked?

third774, 2023-02-20:
I ordered a [Chocofi kit](https://shop.beekeeb.com/product/chocofi-36-keys-kailh-low-profile-choc-v1-mechanical-ergonomic-hotswap-split-keyboard-diy-kit/) that included two [Sea Picro](https://github.com/joshajohnson/sea-picro) microcontrollers and am running into issues getting the firmware installed and working.

When I first plugged them in, my Mac (running 13.1 Ventura) recognized them as a USB storage device. I was trying to use the [QMK toolbox](https://github.com/qmk/qmk_toolbox) to flash the firmware onto them, not realizing that I needed to drag and drop a `.uf2` file instead of trying to flash a `.hex` file.

In the process of figuring that out, I had pressed the reset button a bunch of times and would see a message from MacOS saying I hadn't properly ejected the device, so I was pretty sure that the reset button was working correctly, but just to be sure I tried shorting the reset and ground pins and saw the same behavior.

I decided to just try putting the `.uf2` file on the [docs page here](https://docs.beekeeb.com/chocofi-keyboard) and after I did this for each half of the keyboard, the MacOS keyboard setup prompt did appear, but I was not able to proceed with it because the button to proceed was disabled waiting for me to press a key on the keyboard. I tried pressing keys on the board, but nothing happened. So I closed out of that, and now the Sea Picro no longer shows up as a USB device in Finder.

It does show up in System Information, and when I run:

```sh
> system_profiler SPUSBDataType

USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleT6000USBXHCI

        Corne:

          Product ID: 0x0001
          Vendor ID: 0x4653
          Version: 0.01
          Serial Number: vial:f64c2b3c
          Speed: Up to 12 Mb/s
          Manufacturer: foostan
          Location ID: 0x01100000 / 1
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
```

[Supposedly this board can't be bricked](https://github.com/joshajohnson/sea-picro#features), but it seems to have done it? I've tried:
- Pressing the reset button once
- Pressing the reset button twice quickly
- Holding the reset button for a few seconds
- Holding the reset button for a long time
- Holding the reset button while plugging it in.
- Using different USB ports on my MacBook
- Rebooting
- Using a different MacBook
- Using a Surface Pro 10
- Lots of searching online

Any ideas on what to try next would be super appreciated. I'm also looking into getting openocd set up to see if I can reset the board over USB connection somehow, but running into issues with that as well.

FWIW, I'm pretty confident that I didn't mess up any soldering with the board and header pins so I don't think anything shorted to brick the board apart from me shorting the reset pin, but that seemed to give the same behavior as the reset button, and I was able to put files on it once.

joshajohnson, 2023-02-21:
Hi, 

Given the board enumerates I'd have to assume it's not bricked and instead is not getting into the bootloader.

Holding the reset button down / shorting the RST button to GND whilst inserting should bring the device into it's ROM bootloader.

If that's not happening, can you try shorting the "B" test point on the back of the board to GND whilst plugging it in which skips the reset button to force it into bootloader mode.
 
If the RP2040 hadn't been flashed before you got it, it would first enumerate in the bootloader and allow you to flash the `.uf2` file, but if the reset circuit is not working correctly it would not allow you to get back into the bootloader.

If after trying the above the issue continues, I'd suggest reaching out to the place your purchased Sea-Picro from and ask for confirmation all of the features were tested before shipping to ensure the unit shipped to you isn't defective.

third774, 2023-02-21:
Thanks! Turns out I installed the micro controller facing the wrong direction. Apparently these are the symptoms. 🙈