joshajohnson/sea-picro #16: make new fails

idank, 2023-02-21:
Trying to build the gerbers locally, I'm getting:

```
$ make new
git submodule update --init --recursive --progress
Submodule path 'josh-kicad-lib': checked out 'ef4384a867d9c2ea6f2b9b7983db1512670b1de6'
cd josh-kicad-lib && git checkout master && git pull
Previous HEAD position was ef4384a dfm for sea-picro
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
Already up to date.
cd josh-kicad-lib && bash setup.sh "1.1" "idank" "Josh Johnson"
sed: 1: "../hardware/1.1/idank.k ...": invalid command code .
sed: 1: "../hardware/1.1/idank.k ...": invalid command code .
+++ DONE
```

Did I miss a step? Also I'm no Make expert but where are these rules defined?

```
prod-files:
	make gerb
	make pnp
	make bom
```

Are there ready to go gerbers/cpl files out there so I don't have to build locally?

Thanks!

joshajohnson, 2023-02-21:
That might be old code that I forgot to delete...

Gerbers are in the [releases](https://github.com/joshajohnson/sea-picro/releases) section.

closed 2023-02-21 by joshajohnson

idank, 2023-02-21:
Nice, thanks. Do you happen to also have CPL files for JLC?

idank, 2023-02-21:
Never mind, I converted the .pos file.