joshajohnson/sea-picro #20: I2C woes

idank, 2023-06-13:
Hey Josh,

I've been having some weird issues with a [trackball i2c device](https://shop.pimoroni.com/products/trackball-breakout?variant=27672765038675). I've used dozens of this on generic pro micros just fine, but for some reason it doesn't work quite right with the sea picro.

There's some apparent spikes during big moves as seen here: https://i.imgur.com/9OhksMT.mp4 (starts around the 5 second mark). It doesn't seem to be firmware related (I looked at the raw i2c reads in QMK, they're just bigger when the spikes occur). Nor is the trackball unit itself at fault as it works fine when wired to a pro micro.

The only evidence I've found of this being an issue specific to the RP2040 is https://github.com/arduino/ArduinoCore-mbed/issues/448, but I don't have the tools or knowledge to test this out.

Do you have any thoughts on this? Thanks.

joshajohnson, 2023-06-13:
That's strange behavior, I've used the bare trackball before but not that module.

It would be ideal to get a logic analyser trace of what is going on, my guess is that a read is failing and by the time the next read works it's got a large delta which results in the jumpy behaviour. 

I doubt that Arduino issue is relevant, as it's not used in QMK.

Looking at the example code [here](https://github.com/pimoroni/pimoroni-pico/blob/main/examples/breakout_trackball/demo.cpp), it calls out that I2C needs to be less than 100KHz, I wonder what it's set to in QMK?

Otherwise the next step would be getting an oscilloscope / logic analyser on the bus and seeing that there is decent signal integrity, correct clock speed, and looking at the raw packets flying around.

Sadly purchasing those trackballs in Australia are quite expensive, so it's not something I'll be able to source to assist with debugging.

idank, 2023-06-13:
From the [product page](https://shop.pimoroni.com/products/trackball-breakout?variant=27672765038675):

[image]

and it seems the default in [QMK](https://docs.qmk.fm/#/i2c_driver?id=avr-configuration) is 400k, but even after lowering it to 250k, 100k, I'm still getting spikes. :(

idank, 2023-06-15:
Well this is embarrassing. Apparently when QMK is running in split keyboard mode, it'll check for connectivity with the slave side every few second, and these spikes occur whenever the check is happening. I was testing things out with just one side connected. 🤦

closed 2023-06-15 by idank

joshajohnson, 2023-06-16:
That's not what I expected the issue to be, but nice work figuring it out!