joshajohnson/sea-picro #22: Flash chip doesn't match QMK's default

idank, 2023-09-20:
I recently tried my first PCBA order from JLC and upon flashing QMK realized (with the help of the kind folks on discord) that it's configured for a different flash chip than the one in bom, as seen here: https://docs.qmk.fm/#/platformdev_rp2040?id=rp2040-second-stage-bootloader-selection.

It only takes one define to get working, but figured that you might want to document this or perhaps even change the flash in the bom to W25Q080 so folks don't have to deal with this to begin with, seeing as the picro is most commonly used with QMK.

joshajohnson, 2023-09-20:
Fixed in https://github.com/joshajohnson/joshajohnson.github.io/commit/b7dc95860029295b421f1e27cc69c6d4512357c7

idank, 2023-09-21:
Out of curiosity are the two interchangeable on the sea picro, or would other components need to be changed as well?

joshajohnson, 2023-09-22:
They are interchangeable.